oskarpearson/mmeowlink #6: Iter_pump returns no data with 512

amazaheri, 2016-01-22:
Not a major issue but that is the only command not working, tested with both TI and RL on 512

oskarpearson, 2016-02-09:
Not sure how to debug this without access to a 512 :(

I think we're going to need to add some sort of debugging setting that dumps everything received (without serial numbers) so that we can track this down.

@bewest Does anything spring to mind here that might be causing this?

amazaheri, 2016-02-09:
At this point I consider it low priority as it only affects people using older pump, x12 has other limitation as well :)

amazaheri, 2016-02-20:
https://github.com/bewest/decoding-carelink/commit/539148579f17120333adb9d39bcc7e69bbf5d9d8 fixed this issue!

amazaheri, 2016-02-20:
and https://github.com/bewest/decoding-carelink/commit/45d16d6b4f9762bd14ec3ca4d9b645c77859f812 Thank you @bewest

closed 2016-02-20 by amazaheri