oskarpearson/mmeowlink #9: CRC errors with iter_pump

amazaheri, 2016-02-08:
I noticed that during iter_pump command if pump moves (as with kids it happens all the time) we throw and exception due to CRC mismatch, would be good to have a retry logic.

oskarpearson, 2016-02-09:
Hey there

Yeah - this happens a lot to me too. https://github.com/bewest/decoding-carelink/pull/170 is related.

The problem is that the CRC error is happening at the decocare level, which means we can't trap it for retry at the mmeowlink level.

Once we add the PR fix into decocare, we can catch that error at an openaps level and retry it from there.

Oskar

amazaheri, 2016-02-09:
Great!

oskarpearson, 2016-04-16:
@amazaheri Could you confirm this is resolved with the latest version and close the issue? I think it should be. Thanks!

amazaheri, 2016-04-16:
Indeed, have not seen any CRC for a while

closed 2016-04-16 by amazaheri