oskarpearson/mmeowlink #11: bytearray index out of range rf.py

amazaheri, 2016-02-10:
After upgrading to sbug_rfspy 0.6 , decocare-0.0.22.dev and latest mmeowlink on Ras Pi getting the following exception after successfully enacting the temp.
[image]

I think Scott also had this at one point, this looks different that the issue with commands.py

amazaheri, 2016-02-10:
And with latest ver of mmeowlink:

pump://JSON/set_temp_basal/oref0-predict/enacted.json
oref0-predict/enacted.json  raised  Could not get subg_rfspy state or version. Have you got the right port/device and radio_type?
Traceback (most recent call last):
  File "/usr/local/bin/openaps-report", line 4, in <module>
    __import__('pkg_resources').run_script('openaps==0.0.10', 'openaps-report')
  File "build/bdist.linux-armv7l/egg/pkg_resources/__init__.py", line 735, in run_script
  File "build/bdist.linux-armv7l/egg/pkg_resources/__init__.py", line 1652, in run_script
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.10-py2.7.egg/EGG-INFO/scripts/openaps-report", line 82, in <module>
    app( )
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.10-py2.7.egg/openaps/cli/__init__.py", line 44, in __call__
    self.run(self.args)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.10-py2.7.egg/EGG-INFO/scripts/openaps-report", line 75, in run
    output = app(args, self)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.10-py2.7.egg/openaps/cli/subcommand.py", line 52, in __call__
    return self.method.main(args, app)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.10-py2.7.egg/openaps/reports/invoke.py", line 40, in main
    output = task.method(args, app)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.10-py2.7.egg/openaps/uses/use.py", line 44, in __call__
    self.before_main(args, app)
  File "/usr/local/lib/python2.7/dist-packages/openaps-0.0.10-py2.7.egg/openaps/vendors/medtronic.py", line 61, in before_main
    self.setup_medtronic( )
  File "/home/pi/mmeowlink/mmeowlink/vendors/mmeowlink.py", line 79, in setup_medtronic
    setup_func(self)
  File "/home/pi/mmeowlink/mmeowlink/vendors/mmeowlink.py", line 61, in setup_medtronic_link
    link = LinkBuilder().build(radio_type, port)
  File "/home/pi/mmeowlink/mmeowlink/link_builder.py", line 11, in build
    return SubgRfspyLink(port)
  File "/home/pi/mmeowlink/mmeowlink/vendors/subg_rfspy_link.py", line 40, in __init__
    self.open()
  File "/home/pi/mmeowlink/mmeowlink/vendors/serial_interface.py", line 27, in open
    self.check_setup()
  File "/home/pi/mmeowlink/mmeowlink/vendors/subg_rfspy_link.py", line 45, in check_setup
    self.serial_rf_spy.sync()
  File "/home/pi/mmeowlink/mmeowlink/vendors/serial_rf_spy.py", line 90, in sync
    raise CommsException("Could not get subg_rfspy state or version. Have you got the right port/device and radio_type?")
mmeowlink.exceptions.CommsException: Could not get subg_rfspy state or version. Have you got the right port/device and radio_type?

amazaheri, 2016-02-10:
Installing @ps2 mmeowlink branch fixed the issue!

ps2, 2016-02-10:
The error you were seeing above looks like the RL/cc1110 was in a bad state.  I'm guessing you power cycled, and thats what fixed the error, not my changes.

amazaheri, 2016-02-10:
no it was with TI

amazaheri, 2016-02-10:
and didn't power cycle it :)

amazaheri, 2016-02-10:
was going to dig into code but no need anymore I guess

closed 2016-02-10 by amazaheri