oskarpearson/mmeowlink #14: Setting multiple temp basals

channemann, 2016-03-08:
You can see in the history below (from a 723 v2.4A) that the same temp basal was set three times, with the first two lasting only two seconds. It appears as if this is related to the pump not returning an acknowledgement, and is why, in decocare, the default retries for setting temp basals is zero (https://github.com/bewest/decoding-carelink/blob/a1c9d823cc5357415c89989d9dcdad872c430f51/decocare/commands.py#L259-L267).

This happens on every loop, but does not have an immediate impact on functionality (aside from cluttering the pump history and adding a few seconds to the loop duration).

cc @loudnate @bewest 

```
{
    "_type": "TempBasalDuration", 
    "duration (min)": 30, 
    "_description": "TempBasalDuration 2016-03-07T22:05:02 head[2], body[0] op[0x16]", 
    "timestamp": "2016-03-07T22:05:02", 
    "_body": "", 
    "_head": "1601", 
    "_date": "02c5164710"
  }, 
  {
    "_type": "TempBasal", 
    "temp": "absolute", 
    "_description": "TempBasal 2016-03-07T22:05:02 head[2], body[1] op[0x33]", 
    "timestamp": "2016-03-07T22:05:02", 
    "_body": "00", 
    "_head": "3319", 
    "rate": 0.625, 
    "_date": "02c5164710"
  }, 
  {
    "_type": "TempBasalDuration", 
    "duration (min)": 30, 
    "_description": "TempBasalDuration 2016-03-07T22:05:00 head[2], body[0] op[0x16]", 
    "timestamp": "2016-03-07T22:05:00", 
    "_body": "", 
    "_head": "1601", 
    "_date": "00c5164710"
  }, 
  {
    "_type": "TempBasal", 
    "temp": "absolute", 
    "_description": "TempBasal 2016-03-07T22:05:00 head[2], body[1] op[0x33]", 
    "timestamp": "2016-03-07T22:05:00", 
    "_body": "00", 
    "_head": "3319", 
    "rate": 0.625, 
    "_date": "00c5164710"
  }, 
  {
    "_type": "TempBasalDuration", 
    "duration (min)": 30, 
    "_description": "TempBasalDuration 2016-03-07T22:04:58 head[2], body[0] op[0x16]", 
    "timestamp": "2016-03-07T22:04:58", 
    "_body": "", 
    "_head": "1601", 
    "_date": "3ac4164710"
  }, 
  {
    "_type": "TempBasal", 
    "temp": "absolute", 
    "_description": "TempBasal 2016-03-07T22:04:58 head[2], body[1] op[0x33]", 
    "timestamp": "2016-03-07T22:04:58", 
    "_body": "00", 
    "_head": "3319", 
    "rate": 0.625, 
    "_date": "3ac4164710"
  }, 
```

bewest, 2016-03-08:
This is mainly for note taking and posterity, correct? No harm no foul?

bewest, 2016-03-08:
So question for Medtronic:

Any recommendations on setting temporary basal rates?  Any recommendations to ensure successful delivery of temporary basal rate, recommendations on checking existence of temporary basal rates, any recommendations on speeding up pump history (to double check temporary basal rate is enacted and logged)?

loudnate, 2016-03-08:
Correct, no harm. I just wrote this command in the iOS RL driver and had to set retries to 0, and fetch temp basal status to confirm.

I haven't found a race condition in the final step yet, so I'm not sure speeding up history is a requirement.

oskarpearson, 2016-04-16:
@channemann Could you confirm this is resolved for you with the latest version and close the issue? Thanks!

channemann, 2016-04-16:
@oskarpearson confirmed; thanks for the fixes!

closed 2016-04-16 by channemann