oskarpearson/mmeowlink #16: When using the command "openaps use pump mmtune", the help text does not appear

oskarpearson, 2016-03-17:
Migrating https://github.com/ps2/subg_rfspy/issues/24 to this repo. @andrew-warrington

>The file mmtune.py includes help text explaining required arguments. This text does not appear when using "openaps use pump mmtune" command.

>This may possibly be an openaps topic rather than a mmtune topic.

>Help text:
>print "Usage: mmtune.py /dev/ttyMFD1 pumpserial [radio_locale]"
>print "Radio locale defaults to 'US'. Set to 'WW' for other countries"

oskarpearson, 2016-04-16:
@andrew-warrington - could you include the relevant bit of your openaps config here? I can then test.

The latest dev version of mmeowlink changes the mmtune command to use parameters that should better suit openaps. For eg:

`mmtune.py --port /dev/ttyACM1 --radio_locale WW --serial 123123`

PieterGit, 2016-06-25:
please also document the way to use WW pumps with openaps on  https://github.com/oskarpearson/mmeowlink/wiki/Installing-MMeowlink

I ended up hacking mmtune.py 
i changed US to WW in mmeowlink/mmtune.py
def init(self, link, pumpserial, radio_locale='US'):

is the preferred way: 
openaps use pump mmtune WW
or something else?

PieterGit, 2016-09-24:
It's not only the help text of the mmtune, but I think it's impossible (at least for me) to be able to use mmeowlink with a WW medtronic pump .

I created a pull request for the oref0-setup.sh script, to be able to use mmeowlink with a world wide / European pump. See https://github.com/openaps/oref0/pull/226/files

The work around is: 

```sed -i "s/def __init__(self, link, pumpserial, radio_locale='US'):/def __init__(self, link, pumpserial, radio_locale='WW'):/g" ~/src/mmeowlink/mmeowlink/mmtune.py || die "Couldn't patch mmtune.py"```

I don't think an inline replace with `sed` is a proper fix, but any help with getting this fixed properly is very welcome.

closed 2021-03-22 by oskarpearson