oskarpearson/mmeowlink #22: mmeowlink-bolus.py ack returns false when true

kenstack, 2016-04-16:
boluses are being delivered fine but the ack returns false - Im trying to use the acks for confirmation that bolus was received

{"recieved": false, "_type": "bolusrequest"}

also - i before e except after c ;)

oskarpearson, 2016-04-16:
Hi Ken. Thanks for the report!

Could I ask you to double-check you're on the very latest version of mmeowlink and potentially the development version of decoding-carelink? Obviously you'd need to be pretty careful doing this!

I've unfortunately got no way to test the bolus code at the moment - no spare pump and it's in use continuously. I know that @bewest made some changes in the development version of decoding-carelink recently that changed the send/receive response sizes to better work with mmeowlink. I'm not sure if the Bolus code changed part of that.

The `recieved` vs `received` issue is a longstanding bug in decoding-carelink, so there's no real way for us to change that in mmeowlink. It should probably be logged against the https://github.com/bewest/decoding-carelink/issues repo. (I'm pretty sure there was a bug there for this, but can't seem to track it down.)

Cheers!

Oskar

kenstack, 2016-04-17:
Im on the latest version of mmeow but on master of decoding-carelink.  @bustavo and I are going to do some debugging this week on the responses - will post results and hopefully a PR

oskarpearson, 2016-05-30:
@kenstack Just checking in on this - let me know if it's still a problem and we can re-open :)

closed 2016-05-30 by oskarpearson

bewest, 2016-10-26:
```
GOT MODEL ReadPumpModel:size[64]:data:'522'
Namespace(autoinit=False, dryrun=False, init=True, no_postlude=False, no_prelude=False, no_rf_prelude=False, port='scan', radio_type='subg_rfspy', serial='571347', session_life=10, strokes_per_unit=10, units=0.1, verbose=[1, 1, 1, 1, 1])
DEBUG:mmeowlink.handlers.stick:*** Sending prelude for command 66
DEBUG:mmeowlink.vendors.subg_rfspy_link:write_and_read: a75713474200bf
DEBUG:mmeowlink.vendors.serial_rf_spy:command 5
DEBUG:mmeowlink.vendors.serial_rf_spy:params: 0000000001f400a96956c63d16d325552dc000
DEBUG:mmeowlink.vendors.serial_rf_spy:get_response: timeout = 1.5
DEBUG:mmeowlink.vendors.serial_rf_spy:buf = aa00
ERROR:mmeowlink.handlers.stick:Timed out or other comms error - Received an error response Timeout - retrying: 1 of 3
DEBUG:mmeowlink.handlers.stick:*** Sending prelude for command 66
DEBUG:mmeowlink.vendors.subg_rfspy_link:write_and_read: a75713474200bf
DEBUG:mmeowlink.vendors.serial_rf_spy:command 5
DEBUG:mmeowlink.vendors.serial_rf_spy:params: 0000000001f400a96956c63d16d325552dc000
DEBUG:mmeowlink.vendors.serial_rf_spy:get_response: timeout = 1.5
DEBUG:mmeowlink.vendors.serial_rf_spy:buf = 3f38a96956c63d165665558c
DEBUG:mmeowlink.vendors.serial_rf_spy:buf = 3f38a96956c63d165665558cd500
DEBUG:mmeowlink.handlers.stick:len(params)  == 1
DEBUG:mmeowlink.vendors.subg_rfspy_link:write_and_read: a7571347420101000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000fd
DEBUG:mmeowlink.vendors.serial_rf_spy:command 5
DEBUG:mmeowlink.vendors.serial_rf_spy:params: 0000000001f400a96956c63d16d3257157155555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555555570d000
DEBUG:mmeowlink.vendors.serial_rf_spy:get_response: timeout = 1.5
DEBUG:mmeowlink.vendors.serial_rf_spy:buf = 3f
DEBUG:mmeowlink.vendors.serial_rf_spy:buf = 3f39a969
DEBUG:mmeowlink.vendors.serial_rf_spy:buf = 3f39a96956c63d
DEBUG:mmeowlink.vendors.serial_rf_spy:buf = 3f39a96956c63d165665
DEBUG:mmeowlink.vendors.serial_rf_spy:buf = 3f39a96956c63d165665558cd5
DEBUG:mmeowlink.vendors.serial_rf_spy:buf = 3f39a96956c63d165665558cd500
INFO:decocare.commands:Bolus:size[0]:data:{'recieved': False, '_type': 'BolusRequest'}:download:done?explain=found[64] expected[0]
response: Bolus:size[0]:data:{'recieved': False, '_type': 'BolusRequest'}
```
#### decoded:
```python
{'recieved': False, '_type': 'BolusRequest'}
```