oskarpearson/mmeowlink #37: Please proofcheck external antenna addition to wiki

ktomy, 2016-06-06:
Hi,

I have added external antenna mounting part to https://github.com/oskarpearson/mmeowlink/wiki/Pi-With-Slice-of-Radio but as my English is too poor, please proofcheck my addition. Also I cannot add pictures to the wiki, could you please add some of the photos from [here](https://goo.gl/photos/ou4JGpHMbVUf7GcM8) to the wiki page?

Thank you.

oskarpearson, 2016-09-15:
Hey @ktomy the wiki text looks absolutely perfect to me. And those are great photos!

I'm sorry I'm only getting to this now. I'm not sure how I missed all the github issues/PRs.

Uploading a photo is a bit of a pain. The best suggestion I've seen is to attach them to an issue, like this one, by dragging-and-dropping into the github interface. Then you can link to them by right clicking and getting the URL and pasting into the text. https://help.github.com/articles/adding-images-to-wikis/ has some info on the link formatting. If you run into issues, let me know which ones you'd like where in the text and I'll give it a go. :)

Thanks again!

oskarpearson, 2016-10-10:
Closing this for now so as to reduce clutter. Hope that's ok.

closed 2016-10-10 by oskarpearson