oskarpearson/mmeowlink #42: Add NAK logic to request retransmission from pump based on @ecc1's code

wkcwells, 2016-08-17:
Code is here: https://github.com/ecc1/medtronic

wkcwells, 2016-08-17:
I have a basic prototype working.  Still refining it...

oskarpearson, 2016-09-15:
Heya Keving. I'm just discovering the wonders of the golang code that Eric worked on. I wasn't entirely sure what this issue relates to, though? Would you mind adding some detail/explanation? Thanks!

ecc1, 2016-09-16:
When requesting a history page from a MM pump, the protocol returns 16 records to make up the whole page of data.  The client sends the request, gets the first record, then has to send an ACK for each subsequent record.

The records have sequence numbers 1..16.  Previously, if you were expecting #7 but instead got #8, you would realize that you'd lost #7, throw up your hands, and return failure for the whole operation.

I discovered that the protocol also accepts a NAK instead of an ACK, which will cause retransmission of the *current* record.  Unfortunately, the ACKs and NAKs *don't* have sequence numbers so the logic is a little tricky -- you can't wait until you see an unexpected sequence number because then it's too late.  You have to send the NAK when you timeout expecting a response.

Anyway, the net result is that history page fetches in noisy or marginal comms environments are much more likely to suceed.