oskarpearson/mmeowlink #43: issues with Raspberry pi 3

thebookins, 2016-08-19:
It appears that the fully-featured serial port on the raspberry pi 1 and 2 has been removed from the GPIO pins and a mini UART installed instead (http://raspberrypi.stackexchange.com/questions/45570/how-do-i-make-serial-work-on-the-raspberry-pi3). 

One consequence of this is that you need to select the port /dev/ttyS0 or /dev/serial0. Other than this, should we expect mmeowlink work with the mini UART?

oskarpearson, 2016-09-15:
Hi - have you been successful with this? I don't have a handy way of testing things with the pi3, unfortunately.

It might be best to put something about this on a new page, titled something like "Serial Ports on the Pi 3" - https://github.com/oskarpearson/mmeowlink/wiki/_new We can then link it in the relevant other wiki pages.

Thanks for contributing!

dcacklam, 2018-03-05:
It appears to work just fine with either the mini, or the full UART.

closed 2018-03-22 by oskarpearson