oskarpearson/mmeowlink #51: Easier support for non-USA (WW) pumps

oskarpearson, 2016-10-10:
Currently it's not easy to use non-USA pumps with Mmeowlink and OpenAPS.

At the moment, I do this manually in my loop, by running `mmtune.py --port ${PORT} --serial FIXME --radio_locale WW` outside of OpenAPS. Some hacks have been proposed to deal with this, inclduing https://github.com/openaps/oref0/pull/226#event-809257329

Additionally, some WW pumps require this process, since the default subg radio settings may need some adjustment: https://github.com/oskarpearson/mmeowlink/wiki/Non-USA-pump-settings

It would be better if:
- It was easy to add a new device with 'openaps device add pump mmeowlink' and specify the radio locale at the same time.
- The radio locale would be stored in the pump.ini, and the radio would automatically use it.
- It was easy to integrate custom radio settings into the mmeowlink tool, since radios vary from user to user.
- We should default to the parameters in https://github.com/oskarpearson/mmeowlink/wiki/Non-USA-pump-settings as the majority of WW users need them. (This may need to happen as a Pull Request for https://github.com/ps2/subg_rfspy instead)

oskarpearson, 2016-10-10:
I started doing some work on this at https://github.com/oskarpearson/mmeowlink/tree/radio_parameters - but it needs a lot of work, since it predates many other changes. Structurally, though, I like the way we pass/set parameters.

I'll organise to do a rebuild of that tree ASAP

oskarpearson, 2016-10-10:
Related, and which should probably be fixed as part of this change: https://github.com/oskarpearson/mmeowlink/issues/16

PieterGit, 2016-11-07:
@oskarpearson : can you tell me what needs to be done, and how you think it's the best way to implement it?  I can see if I can help a bit but I don't know what changes you intend to make.

scottleibrand, 2016-12-04:
Anyone have any updates on this?  Wondering if we should go ahead and merge https://github.com/openaps/oref0/pull/232, or if there is a reasonable path forward toward doing this right...

oskarpearson, 2016-12-13:
If anyone wants to give this a test on a new box, this will be fixed with:

https://github.com/openaps/openaps/pull/112
https://github.com/openaps/oref0/pull/276
and the dev branch of mmeowlink.

I'll be doing an end-to-end test shortly.

oskarpearson, 2016-12-13:
I've merged the latest changes into master, and built/published a release in preparation of changes to oref0/openaps: mmeowlink 0.10.0

I'm going to leave this open for now - until any issues with those are resolved.

PieterGit, 2016-12-29:
@oskarpearson i integrated run subg-ww-radio-parameters script on mmtune
with oref0-setup.sh 
Can you review https://github.com/openaps/oref0/pull/282
(especially https://github.com/openaps/oref0/pull/282/commits/bee2e250bac23b3c9c3f29f8a14213236916ba76 )

I still had a problem with the radio_locale, see https://github.com/oskarpearson/mmeowlink/issues/55

closed 2021-03-22 by oskarpearson