oskarpearson/mmeowlink #52: udev symplink problem on pi zero / ti usb not working on pi zero

PieterGit, 2016-11-06:
when i use the udev rules i end up with a symlink like this:
```
pi@pi0$ ls -al /dev/mmeowlink
lrwxrwxrwx 1 root root 15 Nov  6 11:17 /dev/mmeowlink -> bus/usb/001/006
```
instead of a symlink to ttyACM<x>.

PieterGit, 2016-11-06:
Linux kernel 4.4.30+ #919 Tue Nov 1 16:52:58 GMT 2016 armv6l GNU/Linux
TI firmware 0.8
Same TI stick works fine with pi3.
```
pi@pi0:/etc/udev/rules.d $ cat 90-mmeowlink.rules
# symlink for TI Dongle flashed with subg_rfspy firmware
ATTR{idVendor}=="1d50", ATTR{idProduct}=="8001", MODE="0664", GROUP="plugdev", SYMLINK+="mmeowlink", RUN+="/usr/local/bin/mmt.sh"
 ```
the mmt.sh script does a mmtune
```
/usr/local/bin/mmtune.py --subg_rfspy --radio_locale WW --port /dev/ttyACMX --serial 123456 
````

PieterGit, 2016-11-06:
I found a workaround
```
$ cat /etc/udev/rules.d/90-mmeowlink.rules
# symlink for TI Dongle flashed with subg_rfspy firmware
ATTR{idVendor}=="1d50", ATTR{idProduct}=="8001", MODE="0664", GROUP="plugdev", SYMLINK+="mmeowlink", RUN+="/usr/local/bin/mmt.sh"
```

and `$cat /usr/local/bin/mmt.sh` contains:
```
#!/bin/sh
echo `date ` mmtune start >> /var/log/openaps/mmtune.log
echo 'change link from ' >> /var/log/openaps/mmtune.log
ls -al /dev/mmeowlink >> /var/log/openaps/mmtune.log
d=`/home/pi/src/oref0/bin/oref0-find-ti-usb-device.sh`
sudo rm -rf /dev/mmeowlink
echo "to:"
sudo ln -s /dev/$d /dev/mmeowlink
ls -al /dev/mmeowlink  >> /var/log/openaps/mmtune.log
/usr/local/bin/mmtune.py --subg_rfspy --radio_locale WW --port /dev/mmeowlink --serial 123456 >> /var/log/openaps/mmtune.log
echo `date ` mmtune end >> /var/log/openaps/mmtune.log
```

If there is anything I can do to fix this without the mmt.sh script I would be thankfull. Please let me know if I can help to fix this.

I think this also fix the `serial.serialutil.SerialException: Could not configure port: (25, 'Inappropriate ioctl for device')`. For example reported here: https://gitter.im/nightscout/intend-to-bolus?at=57ad0ac3613cc180367ba884 by @bfaloona

oskarpearson, 2016-11-06:
Hi Pieter. Currently worldwide pump support is problematic. The way we do it is that we run mmtune every time openaps runs - rather than in the udev rules. The radio frequency of the pump/ti stick change as temperature changes, for example. By doing things in udev rules, you're only running the tune once - when the ti stick is inserted or when the device is rebooted.

If you change things so that you run mmtune --radio_locale WW as part of your loop it'll re-tune as the day goes by and as the radio frequency drifts.

There's an existing issue for WW pump support here: https://github.com/oskarpearson/mmeowlink/issues/51 If it's ok, I'd like to close this as a duplicate of that item?

I don't think that the 'Inappropriate ioctl for device' issue is related to the WW pump issue. That might be a different bug, though. We'd need more information on where /dev/ttyACM0 or similar pointed at, and how it varied vs /dev/mmeowlink

PieterGit, 2016-11-07:
@oskarpearson  thanks for responding. I don't think this specific issue is WW only. I have seen quite some `serial.serialutil.SerialException: Could not configure port: (25, 'Inappropriate ioctl for device')` reports by people on https://gitter.im/nightscout/intend-to-bolus

Strange thing is that @mawhit reports it on a PI3 and I have no problems with the TI-stick on my PI3, but do have problems with a PI0.

Perhaps @bfaloona and  @mawhit can let us know if they are using a worldwide (WW) or US pump.

mawhit, 2016-11-07:
I'm using a US pump - MMT723