oskarpearson/mmeowlink #53: Would like to discuss best way to read enlite/others sensor signal directly from transmitter through mmeowlink...

bustavo, 2016-11-06:
I've discussed this with @ps2 ... 

For Enlite users like me, sometimes the pump misses sensor signal for 30 min or more while asleep and then pump marks this as sensor signal level error and ignores the sensor from that point on... 

I've found no command to re-connect the sensor through decocare so I was wondering if I could use mmeowlink to listen for sensor signal to use directly instead of extracting from pump.

I was able to do that with @jberian's mmcommander but am not sure how to do this through mmeowlink.

I saw /mmeowlink/mmeowlink/detect_radio_comms.py but was unable to run it:

`
$ python detect_radio_comms.py 
Traceback (most recent call last):
  File "detect_radio_comms.py", line 3, in <module>
    from exceptions import CommsException
ImportError: cannot import name CommsException
`

Any comments or leads appreciated :)

scottleibrand, 2016-11-06:
We fixed that import error in a newer branch of mmeowlink. I believe it was just a one-line add. LMK if you can't find it.

bustavo, 2016-11-06:
@scottleibrand I tried the dev branch but the error is still there

Could that script be used to sniff the enlite?

scottleibrand, 2016-11-06:
https://github.com/scottleibrand/mmeowlink/commit/44dda6078afbb00b75d388c90601c16c4c689a21

No idea.

oskarpearson, 2016-11-06:
Hi Bustavo. Dev definitely does have that patch - could you do a 'git pull' and check everything's up to date?

In terms of dumping traffic, there's an existing 'snoop like' tool that will show you the received packets: bin/mmeowlink-rf-dump.py It uses the core implementation code at mmeowlink/cli/rf_dump_app.py

The any-pump-comms code is in bin/mmeowlink-any-pump-comms.py mmeowlink/cli/any_pump_comms_app.py and mmeowlink/detect_radio_comms.py

You could write a similar implementation in the mmeowlink/cli/ directory and use it the same way if you want. I'm not sure how reconnecting through decocare would work though.

You'd need to put a 'customize_parser' and prelude methods in to your cli/*_app file - calling super and setting args.no_rf_prelude similarly to mmeowlink/cli/rf_dump_app.py - but otherwise the rest should be pretty clean to code.

I've had a theory about lost sensors for a while - since we get them often.

I think because the pump and the sensor transmit on the same frequency (and the sensor just transmits at a fixed time value without retrying), when openaps chats to the pump at the wrong time it corrupts the radio transmission from the sensor. When this happens a few times (just by chance, since the openaps run chats for a while) then the pump loses the sensor.

If you're running the openaps code every 5 mins exactly, and the sensor timing happens to coincide with that, it'll definitely lose sensor, since no data will ever get to the pump. This is one of the reasons I run Siobhan's pump with a random delay between each run.

I've been wanting to experiment with doing things a different way for a while: instead of openaps communicating with the pump every 5 minutes (or whatever), it'd instead run a program  that would wait until it sees some sensor-originated packets. It'll then wait 30-40 seconds, and then kick off openaps.

When OpenAPS completes, it'll then go back to sitting waiting for the next detected sensor packets. This way openaps won't scramble the sensor comms by "talking over the sensor".

bustavo, 2016-11-06:
Ah! I like the idea of waiting for a new sensor value to initiate the loop! 

Regarding the lost sensor... I think I will agree with you on that! There have been times after calibrations where I start getting lots of skipped data points, I notice that those times, the bg times are around the times where the radio comm is active for the pump...

Besides that sensor comm corruption issue, I get lost sensors after sleeping on them for a while at night or if I turn away from my rig... so the pump declares a lost sensor states after around 30-40 minutes of not getting any values from the sensor... I need to get around that and I think getting the data from the sensor directly will help.

Ok, so I'll start running some tests with the mentioned scripts, I'll update if I find any errors and do a pull-request if I get to develop a new script for that.

Thanks!!

bustavo, 2016-11-07:
Just tested /bin/mmeowlink-rf-dump.py and it works. I am able to read packets being transmitted between my rig and the pump. 

I was also able to retrieve packets from what I think are the enlite transmitter comm with the pump...

Now I need to figure out how to decode the information. There are some clues about how to do it on mmcommander...

closed 2017-01-29 by oskarpearson